stereotech/STE-App-Client #197: Нет доступа к флешке через интерфейс STE App

ShvetsAA, 2021-01-21:
Северсталь не подключила принтер к сети, файлы собираются гонять флешкой.

Сейчас, вставляя флешку в принтер, нельзя посмотреть файлы на ней, ни с экрана принтера, ни через браузер.

frylock34, 2021-01-21:
1. Файлы должны находиться в корне флешки
2. Нужно попробовать включить принтер с вставленной флешкой
3. Добавить в чеклист тестирование печати с флешки

frylock34, 2021-01-21:
Проверить подключена ли флешка в октопринте

ShvetsAA, 2021-01-21:
"Нужно попробовать включить принтер с вставленной флешкой" - выполнено. Флешки в октопринте нет.
[image]

ShvetsAA, 2021-01-21:
При подключении флешки к W5K (Северсталь) происходит перезагрузка RaspberryPi. (2 раза подряд так было, взято со слов Антона, подсветка не гасла при подключении, значит плата/блок питания не участвуют в этом)

ShvetsAA, 2021-01-21:
При подключении к 5BP, светодиод флешки мигает так же как и при подключении к компьютеру (на всякий случай).

frylock34, 2021-03-15:
Исправлен способ монтирования USB накопителей в системе

closed 2021-03-15 by frylock34