stereotech/STE-App-Client #225: При занулении ПГ в Spiral врезается в лицевую панель

frylock34, 2021-03-01:
**Describe the bug**
Неправильно прописываются офсеты в нулевое положение

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform(please complete the following information):**
 - OS: [e.g. iOS, Android, Windows, or Screen Application]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 1.2.0]

**Additional context**
Add any other context about the problem here.

frylock34, 2021-03-04:
Неправильные оффсеты прописываются,  если они нулевые (проверить)

ShvetsAA, 2021-03-04:
Send: M503
Recv: ; config override present: /sd/config-override
Recv: ;Steps per unit:
Recv: M92 X640.00000 Y640.00000 Z800.00000 A40.00000 B40.00000 C80.00000
Recv: ;Acceleration mm/sec^2:
Recv: M204 S1000.00000 Z200.00000
Recv: ;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:
Recv: M205 X0.00100 Z0.00100 S0.00000
Recv: ;Max cartesian feedrates in mm/sec:
Recv: M203 X200.00000 Y200.00000 Z25.00000 S-1.00000
Recv: ;Max actuator feedrates in mm/sec:
Recv: M203.1 X150.00000 Y150.00000 Z25.00000 A120.00000 B120.00000 C120.00000
Recv: ;WCS settings
Recv: G54
Recv: G10 L2 P2 X109.000000 Y39.900002 Z-126.000000 ; G55
Recv: G10 L2 P3 X109.000000 Y126.000000 Z39.900002 ; G56
Recv: ;E Steps per mm:
Recv: M92 E140.0000 P57988
Recv: ;E Filament diameter:
Recv: M200 D0.0000 P57988
Recv: ;E retract length, feedrate:
Recv: M207 S3.0000 F2700.0000 Z0.0000 Q6000.0000 P57988
Recv: ;E retract recover length, feedrate:
Recv: M208 S0.0000 F480.0000 P57988
Recv: ;E acceleration mm/sec��:
Recv: M204 E500.0000 P57988
Recv: ;E max feed rate mm/sec:
Recv: M203 E50.0000 P57988
Recv: ;E Steps per mm:
Recv: M92 E140.0000 P39350
Recv: ;E Filament diameter:
Recv: M200 D0.0000 P39350
Recv: ;E retract length, feedrate:
Recv: M207 S3.0000 F2700.0000 Z0.0000 Q6000.0000 P39350
Recv: ;E retract recover length, feedrate:
Recv: M208 S0.0000 F480.0000 P39350
Recv: ;E acceleration mm/sec��:
Recv: M204 E500.0000 P39350
Recv: ;E max feed rate mm/sec:
Recv: M203 E50.0000 P39350
Recv: ;PID settings, i_max, max_pwm:
Recv: M301 S0 P15.0000 I0.5000 D25.0000 X170.0000 Y170
Recv: ;Max temperature setting:
Recv: M143 S0 P300.0000
Recv: ;PID settings, i_max, max_pwm:
Recv: M301 S1 P15.0000 I0.5000 D25.0000 X170.0000 Y170
Recv: ;Max temperature setting:
Recv: M143 S1 P300.0000
Recv: ;Home offset (mm):
Recv: M206 X0.00 Y0.00 Z0.00 A3.98
Recv: ;Probe feedrates Slow/fast(K)/Return (mm/sec) max_z (mm) height (mm) dwell (s):
Recv: M670 S5.00 K100.00 R0.00 Z200.00 H5.00 D0.00
Recv: ;Home: true, big: 27.800, small: 22.200
Recv: Probe point 0: 100.000,140.000,130.000Probe point 1: 100.000,90.000,130.000Probe point 2: 100.000,140.000,130.000Probe point 3: 150.000,140.000,130.000Probe point 4: 100.000,140.000,130.000Probe point 5: 100.000,140.000,130.000Probe point 6: 100.000,90.000,130.000Probe point 7: 100.000,83.000,145.000Probe point 8: 0.000,0.000,0.000Probe point 9: 0.000,0.000,0.000;Load saved calibration
Recv: M375
Recv: 
Recv: ok
Send: M105

ShvetsAA, 2021-03-04:
После зануления через G10 и установки нулевой точки к основанию в Spiral

frylock34, 2021-03-04:
Подтверждается, будем исправлять