stereotech/STE-App-Client #229: Сброс кнопки пределов на значение по умолчанию "10" при перевороте дисплея телефона с альбомного на книжный и наоборот

ShvetsAA, 2021-03-03:
В менеджере установки нулевой точки, на экране с кнопками правления, при выборе предела, например 1 мм, и при изменении ориентации устройства, установленное значение сбрасывается на стандартное, 10 мм. Глеб уже два раза так врезался.
Приложение до переворота:
[image]
Приложение после поворота:
[image]
Веб версия до поворота:
[image]
Веб версия после поворота:
[image]
Принтер - 3SH, версия прошивки - 2.1.11.DEVELOP

И еще заметил что сильно уезжает разметка этих кнопок, на скринах видно.

Как вариант лечения сброса при перевороте телефона - установить по умолчанию либо 0.1, либо 1 мм. Последствия удара будут явно меньше. Движение по 10 и 100 мм - это осознанный шаг человека, который понимает всю ответственность своего решения, поэтому пусть выбирает каждый раз заново (может быть 100 мм сделать через подтверждение? :)

frylock34, 2021-03-03:
@ShvetsAA На будущее, делайте репорты ТОЛЬКО по веб версии т.к. приложение для андроида м.б. не актуальной версии

ShvetsAA, 2021-03-03:
> @ShvetsAA На будущее, делайте репорты ТОЛЬКО по веб версии т.к. приложение для андроида м.б. не актуальной версии

Принято!

commit 369299b5 (2021-03-15): closes #229
closed 2021-03-15 by frylock34